NVIDIA/open-gpu-kernel-modules #913: Video signal is sometime lost when using VRR with Wayland

thesword53, 2025-08-05:
### NVIDIA Open GPU Kernel Modules Version

575.64.05

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.15.9-arch1-1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GeForce RTX 2080 SUPER

### Describe the bug

When using VRR with the Alienware AW2725DF (2560x1440 360Hz DisplayPort) monitor and the frame rate is low, the monitor sometime looses video signal. I have to disable VRR by closing or unfocusing the game/app to get video signal back. If VRR is switch from on to off, the screen will briefly looses signal. The issue doesn’t happen with my 2 other screens (144Hz and 75Hz).

### To Reproduce

Launch `vrrtest` and close it.
In KDE Plasma display settings set Adaptive Sync to "Always" and wait.

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21607359/nvidia-bug-report.log.gz)

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21607359/nvidia-bug-report.log.gz)

### More Info

The issue can happens in games if the framerate goes below the monitor minimum refresh rate (48Hz) with both KDE and Gnome Wayland.

Monitor EDID: [aw2725df_edid.gz](https://github.com/user-attachments/files/21607236/aw2725df_edid.gz)

Binary-Eater, 2025-08-09:
Thanks @thesword53 for both the log and the EDID. Reading the monitor mode timings and EDID blocks, I am curious about one thing if its fast for you to run.

If you enable VRR (AdaptiveSync), launch `vrrtest`, and set the `vrrtest` fps to 200 fps while fullscreened, do you experiencing monitor blanking?

thesword53, 2026-10-09:
Any update? The issue is still present in the latest drivers (615.78.08).