NVIDIA/open-gpu-kernel-modules #923: Should bad applications be allowed to crash the GPU driver?

birdie-github, 2025-09-03:
### NVIDIA Open GPU Kernel Modules Version

Any

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Any

### Kernel Release

Any

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

Any

### Describe the bug

This shader temporarily crashes the NVIDIA driver:

https://www.shadertoy.com/view/WfSyz3

The system survives but for a few seconds your desktop freezes.

### To Reproduce

Open the URL.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

```
NVRM: Xid (PCI:0000:09:00): 109, pid=32257, name=Renderer, channel 0x00000015, errorString CTX SWITCH TIMEOUT, Info 0x4c01a
```

Binary-Eater, 2025-09-07:
>The system survives but for a few seconds your desktop freezes.

I think that is expected behavior with the example shader program you shared.

>NVRM: Xid (PCI:0000:09:00): 109, pid=32257, name=Renderer, channel 0x00000015, errorString CTX SWITCH TIMEOUT, Info 0x4c01a

Typically, it's better to provide the full nvidia-bug-report.sh logs. However, using just the log you shared this is not a "crash" but a handled exception message. The GPU detected that the graphics channel was being hanged and performed recovery to keep the system in a working state, which is why the desktop recovers. This behavior, I believe, is working as intended.

birdie-github, 2025-09-07:
> I think that is expected behavior with the example shader program you shared.
>
>
> Typically, it's better to provide the full nvidia-bug-report.sh logs. However, using just the log you shared this is [...]

The program crashes every GPU out there with the same error message under any display server (Xorg or Wayland).

I talked to an AI, and it said that this behavior is expected. However, the AI also said that the NVIDIA GPU compiler could be made more verbose so users would understand that they ran a GPU bomb.

Anyway, this error is probably outside the scope of the open-source driver, so let's just close it.

closed 2025-09-07 by birdie-github