NVIDIA/open-gpu-kernel-modules #933: cannot enable HDR on RTX 5060

Zhu-Aemon, 2025-09-13:
### NVIDIA Open GPU Kernel Modules Version

580.82.09-2

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Description:    Arch Linux

### Kernel Release

Linux astraTZ 6.16.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 11 Sep 2025 17:42:36 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 5060 (UUID: GPU-c3ff4948-5d54-0f6c-6950-5e78c6b495eb)

### Describe the bug

There is no way to enable HDR on RTX 5060. I tried different desktop environments, from Gnome to Hyprland to KDE Plasma, HDR does not work on all of them. But when I switch my GPU from 5060 to 3060, HDR works perfect on my platform and settings.
Since proprietary driver does not support 50 series, I figured this is the only place that I can be able to get some help. I am just wondering if HDR is already supported on 5060 by open kernel modules? If so, there must be something very wrong with my configuration. 
If someone is able to get HDR enabled on 5060, please reply to let me know that I am missing something. Your attention is greatly appreciated.

### To Reproduce

Install Arch Linux on a platform with RTX 5060. Install whatever DE you like. Install nvidia-open and whatever packages that are required. Btw I don't think this problem is arch-specific.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/22311599/nvidia-bug-report.log.gz)

Zhu-Aemon, 2025-10-18:
It's been over a month and there still hasn't been any reply to this issue.

Binary-Eater, 2025-10-26:
@Zhu-Aemon sorry for the delay in response. I will try to reproduce this issue early next week and investigate.

Zhu-Aemon, 2026-01-15:
After some tweaking under hyprland. I am able to get HDR working. But the brightness will only appear normal when I manually set `sdr_brightness` to 1.5. I thought this might be a Hyprland issue, might not be related to the driver.
However, HDR still doesn't work under Gnome even though the HDR toggle is on. This also might be a Gnome issue.
This is the EDID information of my monitor
```
00 ff ff ff ff ff ff 00 31 03 59 90 00 00 00 00
10 22 01 03 80 3c 22 78 2b 08 15 ae 50 34 bd 26
0c 50 54 a5 4b 00 81 80 01 01 b3 00 d1 c0 71 4f
a9 c0 81 40 81 c0 08 e8 00 30 f2 70 5a 80 b0 58
6a 00 54 4f 21 00 00 1a 00 00 00 ff 00 30 30 30
30 30 30 30 30 30 30 30 30 30 00 00 00 fd 00 30
a0 1e ff 95 00 0a 20 20 20 20 20 20 00 00 00 fc
00 50 32 37 35 4d 56 0a 20 20 20 20 20 20 02 31

02 03 53 73 4d 01 02 03 04 3f 90 12 13 1f 61 76
5f 11 23 09 07 07 83 01 00 00 67 03 0c 00 10 00
38 42 6d d8 5d c4 01 78 80 63 1a 30 a0 c1 64 06
74 1a 00 00 03 01 30 a0 00 60 00 00 00 00 a0 00
00 00 00 00 00 e3 05 c3 01 e6 06 07 01 8f 89 21
e2 00 7f 6a 5e 00 a0 a0 a0 29 50 30 20 5a 00 54
4f 21 00 00 1a 6f c2 00 a0 a0 a0 55 50 30 20 35
00 54 4f 21 00 00 1a 00 00 00 00 00 00 00 00 91

70 12 79 03 00 03 01 28 48 0e 02 04 ff 0e 7f 01
2f 80 1f 00 6f 08 36 00 02 00 04 00 a0 41 02 04
ff 0e 7f 01 2f 80 1f 00 6f 08 1b 00 02 00 04 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 90

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: LHC
    Model: 36953
    Made in: week 16 of 2024
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 60 cm x 34 cm
    Gamma: 2.20
    DPMS levels: Off
    RGB color display
    First detailed timing is the preferred timing
    Supports GTF timings within operating range
  Color Characteristics:
    Red  : 0.6796, 0.3125
    Green: 0.2050, 0.7382
    Blue : 0.1484, 0.0478
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    DMT 0x06:   640x480    75.000000 Hz   4:3     37.500 kHz     31.500000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0b:   800x600    75.000000 Hz   4:3     46.875 kHz     49.500000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x12:  1024x768    75.028582 Hz   4:3     60.023 kHz     78.750000 MHz
    DMT 0x24:  1280x1024   75.024675 Hz   5:4     79.976 kHz    135.000000 MHz
  Standard Timings:
    DMT 0x23:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz
    DMT 0x3a:  1680x1050   59.954250 Hz  16:10    65.290 kHz    146.250000 MHz
    DMT 0x52:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz
    DMT 0x15:  1152x864    75.000000 Hz   4:3     67.500 kHz    108.000000 MHz
    DMT 0x53:  1600x900    60.000000 Hz  16:9     60.000 kHz    108.000000 MHz (RB)
    DMT 0x20:  1280x960    60.000000 Hz   4:3     60.000 kHz    108.000000 MHz
    DMT 0x55:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz
  Detailed Timing Descriptors:
[... 95 lines cut ...]
  Detailed Timing Descriptors:
    DTD 2:  2560x1440   60.000199 Hz  16:9     88.860 kHz    241.700000 MHz (596 mm x 335 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    5 Vsync  10 Vback   26 Vpol N
    DTD 3:  2560x1440  119.997589 Hz  16:9    182.996 kHz    497.750000 MHz (596 mm x 335 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   5 Vback   77 Vpol N
Checksum: 0x91  Unused space in Extension Block: 8 bytes

----------------

Block 2, DisplayID Extension Block:
  Version: 1.2
  Extension Count: 0
  Display Product Type: Standalone display device
  Video Timing Modes Type 1 - Detailed Timings Data Block:
    DTD:  3840x2160  144.000316 Hz  16:9    318.961 kHz   1347.290000 MHz (aspect 16:9, no 3D stereo)
               Hfront   48 Hsync  32 Hback  304 Hpol P
               Vfront    3 Vsync   5 Vback   47 Vpol N
    DTD:  3840x2160  159.999143 Hz  16:9    350.078 kHz   1478.730000 MHz (aspect 16:9, no 3D stereo)
               Hfront   48 Hsync  32 Hback  304 Hpol P
               Vfront    3 Vsync   5 Vback   20 Vpol N
  Checksum: 0x02
Checksum: 0x90
```